Anime4000/RTL960x #319: Use of RTL960x for Fiber Tapping

LoloBee, 2024-09-30:
Hello!
I'm planning to buy a [profitap bidi device](https://www.profitap.com/1-link-40g-bidi-fiber-tap/) to analyze GPON traffic from the ONU to the ISP.
I need an adapter to receive the mirrored fiber, which is GPON. With one of the suggested SFP adapters and the corresponding firmware, **without** setting the PLOAM password, would the adapter discard the packets, or will it be able to receive them, so I will take care of decryption at software side?

If I'm tracing the path the wrong way, please feel free to correct me.

Thanks.

Anime4000, 2024-10-01:
think link you send is not for GPON, or I been showed incorrectly, it use normal 850nm meanwhile GPON use 1310nm

LoloBee, 2024-10-02:
Sorry I meant [this one](https://www.profitap.com/wp-content/uploads/BiDi-PON-Datasheet.pdf)