Anime4000/RTL960x #320: VSOL V2802RH O5 without WAN IP

jaycelq, 2024-10-05:
My service provider is Singtel in Singapore. I tried to replace the Nokia G240E provided by Singtel with the VSOL V2802RH. I followed the online tutorials for setup, and the PON state changed to O5, but I am unable to obtain a WAN IP. Singtel uses VLAN 10 for Internet and VLAN 30 for VOIP. I saw other issues and tried to set the EntityID 0xe102 0xe10b FwdOp to 0x02, but it failed. Below are my 84 and 171. I am not an expert in PON networks, but from my research, I suspect there is an incompatibility between Nokia's OMCI and VSOL. I would like to seek advice on which direction I should try next.

```
AP#omcicli mib get 84
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX

VlanTagFilterData

XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX

=================================

EntityID: 0xe102

FilterTbl[0]: PRI 0,CFI 0, VID 10

FwdOp:  0x10

NumOfEntries: 1

=================================

=================================

EntityID: 0xe104

FilterTbl[0]: PRI 0,CFI 0, VID 30

FwdOp:  0x10

NumOfEntries: 1

=================================

=================================

EntityID: 0xe10b

FilterTbl[0]: PRI 0,CFI 0, VID 10

FilterTbl[2]: PRI 0,CFI 0, VID 30

FwdOp:  0x10

NumOfEntries: 2

=================================

```

```
AP#omcicli mib get 171
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX

ExtVlanTagOperCfgData

XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX

=================================

EntityId: 0xe01

AssociationType: 10

ReceivedFrameVlanTagOperTableMaxSize: 0

InputTPID: 0x8100

OutputTPID: 0x8100

DsMode: 0

ReceivedFrameVlanTaggingOperTable

INDEX 0

Filter Outer   : PRI 8,VID 30, TPID 5

Filter Inner   : PRI 8,VID 4096, TPID 0, EthType 0x00

Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1

Treatment Inner   : PRI 9,VID 30, TPID 3

INDEX 1

Filter Outer   : PRI 15,VID 4096, TPID 0

Filter Inner   : PRI 8,VID 30, TPID 5, EthType 0x00

Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1

Treatment Inner   : PRI 8,VID 30, TPID 2

INDEX 2

Filter Outer   : PRI 8,VID 10, TPID 5

Filter Inner   : PRI 8,VID 4096, TPID 0, EthType 0x00

Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1

Treatment Inner   : PRI 9,VID 10, TPID 3

INDEX 3

Filter Outer   : PRI 15,VID 4096, TPID 0

Filter Inner   : PRI 8,VID 10, TPID 5, EthType 0x00

Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1

Treatment Inner   : PRI 8,VID 10, TPID 2

INDEX 4

Filter Outer   : PRI 15,VID 4096, TPID 0

Filter Inner   : PRI 15,VID 0, TPID 5, EthType 0x00

Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3

Treatment Inner   : PRI 15,VID 0, TPID 2

INDEX 5

Filter Outer   : PRI 14,VID 4096, TPID 5

Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00

Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3

Treatment Inner   : PRI 15,VID 4096, TPID 3

INDEX 6

Filter Outer   : PRI 15,VID 4096, TPID 0

Filter Inner   : PRI 14,VID 4096, TPID 5, EthType 0x00

Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3

Treatment Inner   : PRI 15,VID 4096, TPID 2

AssociatedMePoint: 0x601

DscpToPbitMapping:

        0x000000

        0x249249

        0x492492

        0x6db6db

        0x924924

        0xb6db6d

        0xdb6db6

        0xffffff

Parsed DscpToPbitMapping:

dscp: 0 => pbit: 0

dscp: 1 => pbit: 0

dscp: 2 => pbit: 0

dscp: 3 => pbit: 0

dscp: 4 => pbit: 0

dscp: 5 => pbit: 0

dscp: 6 => pbit: 0

dscp: 7 => pbit: 0

dscp: 8 => pbit: 1

dscp: 9 => pbit: 1

dscp: 10 => pbit: 1

dscp: 11 => pbit: 1

dscp: 12 => pbit: 1

dscp: 13 => pbit: 1

dscp: 14 => pbit: 1

dscp: 15 => pbit: 1

dscp: 16 => pbit: 2

dscp: 17 => pbit: 2

dscp: 18 => pbit: 2

dscp: 19 => pbit: 2

dscp: 20 => pbit: 2

dscp: 21 => pbit: 2

dscp: 22 => pbit: 2

dscp: 23 => pbit: 2

dscp: 24 => pbit: 3

dscp: 25 => pbit: 3

dscp: 26 => pbit: 3

dscp: 27 => pbit: 3

dscp: 28 => pbit: 3

dscp: 29 => pbit: 3

dscp: 30 => pbit: 3

dscp: 31 => pbit: 3

dscp: 32 => pbit: 4

dscp: 33 => pbit: 4

dscp: 34 => pbit: 4

dscp: 35 => pbit: 4

dscp: 36 => pbit: 4

dscp: 37 => pbit: 4

dscp: 38 => pbit: 4

dscp: 39 => pbit: 4

dscp: 40 => pbit: 5

dscp: 41 => pbit: 5

dscp: 42 => pbit: 5

dscp: 43 => pbit: 5

dscp: 44 => pbit: 5

dscp: 45 => pbit: 5

dscp: 46 => pbit: 5

dscp: 47 => pbit: 5

dscp: 48 => pbit: 6

dscp: 49 => pbit: 6

dscp: 50 => pbit: 6

dscp: 51 => pbit: 6

dscp: 52 => pbit: 6

dscp: 53 => pbit: 6

dscp: 54 => pbit: 6

dscp: 55 => pbit: 6

dscp: 56 => pbit: 7

dscp: 57 => pbit: 7

dscp: 58 => pbit: 7

dscp: 59 => pbit: 7

dscp: 60 => pbit: 7

dscp: 61 => pbit: 7

dscp: 62 => pbit: 7

dscp: 63 => pbit: 7

=================================
```

Anime4000, 2024-10-06:
your ISP using VEIP, which hard to get bridged

jaycelq, 2024-10-07:
> your ISP using VEIP, which hard to get bridged

Thank you very much for your response and help. I have a question: is it because the VSOL 4802RH doesn't support VEIP, or is it due to incorrect configuration? I saw on Hardwarezone that someone succeeded with a Nokia G-010S-A + Media converter. Could you please explain the difference between it and the 4802RH?
[https://forums.hardwarezone.com.sg/threads/whizcomms-new-fibre-broadband-provider.5494964/page-116](url)

d5aqoep, 2024-10-23:
There is a good chance that it is a VLAN issue which happens when OLT is ALCL. You are not using the correct VLAN.
I faced similar problem with Tata Play Fiber. Same ALCL OLT and Nokia Branded Gateway. 

What helped me was:
Clone as much details of your existing ISP gateway like Vendor 4 characters, HW and SW versions. Put VSOL into PPPoE mode and then run the 84 and 171 telnet commands.  I forund out that disabling Bridge Mode temporarily and changing to PPPoE allowed me to get the correct random VLAN which the OLT assigns you.

jaycelq, 2024-10-27:
> @jaycelq There is a good chance that it is a known VLAN issue which happens when OLT is ALCL. You are not using the correct VLAN. I faced similar problem with Tata Play Fiber. Same ALCL OLT and Nokia Branded Gateway.
> 
> What helped me was: Clone as much details of your existing ISP gateway like Vendor 4 characters, HW and SW versions. Put VSOL into PPPoE mode and then run the 84 and 171 telnet commands. I forund out that disabling Bridge Mode temporarily and changing to PPPoE allowed me to get the correct random VLAN which the OLT assigns you.

I followed the instructions in the thread I metioned use Nokia G-010S-A + Media converter, It works. I also set vlan 10. Not clear about why GPON stick works but VSOL not works. Maybe because GPON stick is also from Nokia.

closed 2024-12-29 by Anime4000