Anime4000/RTL960x #321: Emulation of POTS presence

timetragit, 2024-10-06:
According to https://github.com/Anime4000/RTL960x/blob/main/Docs/OMCI_CLI.md , there is interesting information in OMCI MIB:
1) Cardholder. POTS card presence may be configured here
2) Circuit Pack. Can also contain information about embedded POTS card
3) POTS UNI. Also something about POTS presence

Question: Is there a known way how to setup POTS emulation in this SFP ONU (setup correct MIB table sections)? It seems to be no "FLASH SET" variable which can configure that. Otherwise, my ISP's OLT doesn't provide voip VLAN (and even voip gem too)
Any ideas?

Anime4000, 2024-10-06:
most POST is provisioned by VEIP,
in my country, it serve as VEIP, with the stick, it should bridge VEIP just like PPTP, perhaps try erase mtd3 and try again, it worked for me without needing to emulate POTS

some OLT insist to have POTS, once failed to query, OLT wont bridge the VEIP for POTS, for this rather limited :(

timetragit, 2024-10-07:
Thank you for reply. Is it necessary to erase mtd3? I found manual how to  enable VEIP: https://github.com/Anime4000/RTL960x/blob/main/Docs/VEIP.md
Is there a reason to apply this configuration first?