Anime4000/RTL960x #322: 2.5G issue

dlfrks, 2024-10-14:
Hi. I replaced my ISP ONT to FS GPON-ONU-34-20BI to use 2.5G.
However, the speed only reaches up to exactly 1Gbps.
To check if my plan is really 2.5G, I connected two PCs to the ISP ONT and tested the speed and it was okay.

SFP cage is CRS305-1G-4S+ and link status seems to no problem.
```
[admin@MikroTik] /interface/ethernet> monitor sfp-sfpplus1  
                    name: sfp-sfpplus1
                  status: link-ok
        auto-negotiation: disabled
                    rate: 2.5Gbps
             full-duplex: yes
         tx-flow-control: yes
         rx-flow-control: yes
               supported: 10M-baseT-half,10M-baseT-full,100M-baseT-half,100M-baseT-full,1G-baseT-half,1G-baseT-full,1G-baseX,2.5G-baseT,2.5G-baseX,5G-baseT,10G-baseT,10G-baseSR-LR,10G-baseCR
           sfp-supported: 1G-baseX
```

What should I check?

Anime4000, 2024-10-14:
Mikrotik need fiber to be connected able to access WebGUI, unless you update to ROS 7.16 and tick Ignore RX Loss

dlfrks, 2024-10-14:
> Mikrotik need fiber to be connected able to access WebGUI, unless you update to ROS 7.16 and tick Ignore RX Loss

Internet access works fine, the only problem is that the speed is only 1G.

Anime4000, 2024-10-14:
I try help you here, as the stick are you using is not related to RTL960x family,

but it show as `rate: 2.5Gbps`

can you telnet/ssh to verify `onu lanpsg 0` command
[image]

dlfrks, 2024-10-14:
```
root@SFP:/# onu lanpsg 0
errorcode=0 pport=0 mode=15 enable=1 link_status=5 phy_duplex=1
root@SFP:/#
```

the mikrotik and the stick report 2.5G, but in reality only getting 1G.

Anime4000, 2024-10-16:
both stick and mikrotik connected at 2.5G, are you sure that your internet is 2Gbps?

dlfrks, 2024-10-16:
> both stick and mikrotik connected at 2.5G, are you sure that your internet is 2Gbps?

When I measured on ISP ONT with two PCs, it was 2Gbps.
It was also 2Gbps ​​when the ISP engineer measured it when installing the ONT.

Anime4000, 2024-10-17:
oh, have you tried change the MIB file?

try ssh into and see `/etc/mibs/` then `ls -l`, find a file named `data_1g_8q_us1280_ds512.ini`

if it's there, then try enter this command `fw_setenv mib_file data_1g_8q_us1280_ds512.ini`

this use GEM and T-CONT for above 1Gbps

dlfrks, 2024-10-17:
> oh, have you tried change the MIB file?
> 
> try ssh into and see `/etc/mibs/` then `ls -l`, find a file named `data_1g_8q_us1280_ds512.ini`
> 
> if it's there, then try enter this command `fw_setenv mib_file data_1g_8q_us1280_ds512.ini`
> 
> this use GEM and T-CONT for above 1Gbps

I already tried it but it doesn't work.

Anime4000, 2024-10-17:
Like this you need to edit the MIB file and match the GEM and T-CONT from original ONT, is only way

or you buy another stick that not using Lantiq/Intel SoC

dlfrks, 2024-10-17:
> Like this you need to edit the MIB file and match the GEM and T-CONT from original ONT, is only way

How can I extract it from the original ONT?

>or you buy another stick that not using Lantiq/Intel SoC

I already ordered ODI DFP-34X-2C3 and awaiting for shipping.

Anime4000, 2024-10-17:
> How can I extract it from the original ONT?

Like my ISP, have SSH and Telnet, can access OMCI GET INFO, do this will be waste of time to figure it out

> I already ordered ODI DFP-34X-2C3 and awaiting for shipping.

dlfrks, 2024-10-21:
After changing to DFP-34X-2C3, the problem was solved. thanks!

closed 2024-10-21 by dlfrks