Anime4000/RTL960x #324: ODI DFP-34X-2C2 How to update WEB UI?

billbarni, 2024-10-18:
I've tried running the emulator and compying the /Firmware_Mod/DFP-34X-2C2 to /custom, the files apparently get copied to the firmware, and it's packed.
I upload the firmware using the old UI.
When the DFP-34X-2C2 reboots, old UI appears instead of new one.

How to make sure the UI is updated? Any other step is necessary?

Anime4000, 2024-10-19:
the `custom` folder is a mirror or overlay of `squashfs-root`, place it accordingly
https://github.com/Anime4000/RTL960x/tree/main/Tools/emulator#folder-structure

if you using my V2801F as template, make sure ASP tag id it correct

billbarni, 2024-10-19:
I did that. I followed the instructions. I even checked the folder after build, the files are there correctly with the javascript tag. I even changed the firmware used from 0 to 1.

Anime4000, 2024-10-19:
try run browser with private mode to avoid being cache?

billbarni, 2024-10-19:
Not working.

1 - Open terminal
2 - Navigate to emulator folder, run the emulator
3 - Open and copy /Firmware_Mod/DFP-34X-2C2/* to /Tools/emulator/custom/
4 - Exit emulator
5 - Upload new .tar file via GUI

Am I missing any step?

Anime4000, 2024-10-19:
all required bin as fulfilled?

qemu, binwalk, squashfs

also, run qemu-test.sh along side with firmware as per Help says

./qemu-test.sh firmware.bin

you will see a busybox prompt

the custom folder is the mirror of rootfs

try replace logo first,

V2801F asp file are not compatible with ODI, like I said, both using different ASP Tag Id

billbarni, 2024-10-19:
I have a DFP-34X-2C2.
I copied several different DFP-34X-2C2 firmwares. None work.
Thats ok. I cant get it to work anyways. 
ocmicli mib get 84 and 171 return XXXXXX
Web GUI update is being done correclty by me and does NOT work.

billbarni, 2024-10-20:
@Anime4000 just so I can understand.
I cannot use the /web/ directory from V2801F?
DFP-34X-2C2 is not upgradeable? Since there is no /web/ directory for in in the /Firmware_Mod/DFP-34X-2C2/ section?
What TAG should be changed?

Anime4000, 2024-10-20:
for that, you need refer back original asp file and update them

billbarni, 2024-10-20:
Is there any documentation on how to do this?
Sorry to keep bothering about this.

Anime4000, 2024-10-20:
No, it should be easy when you see the HTML with ASP Tag

`<% properties %>`

closed 2024-10-20 by billbarni