Anime4000/RTL960x #325: ODI DFP-34X-2C3 unusual received packets

rapdodge, 2024-10-19:
Hello there, I'm new to using the XPON ONU SFP. I'm using a media converter and I've seen there is unusual packets that received by my router

[image]

My ISP is using VLAN 406, and I don't know why the eth0 is having very much receive packets and the vlan 406 even still not reach half of it

Is there anyone have idea about that?

Best regards,

rapdodge, 2024-10-19:
This is the `bmon -p eth0`

[image]

This is the `iftop -i eth0`

[image]

This is the `ip ro`

[image]

Anime4000, 2024-10-19:
this could be IPTV ?

rapdodge, 2024-10-19:
> this could be IPTV ?

Maybe, but I didn't assign any VLAN to it...

[image]

How to eliminate it?

vuducdong, 2024-10-19:
> > this could be IPTV ?
> 
> 
> 
> Maybe, but I didn't assign any VLAN to it...
> 
> 
> 
> [image]
> 
> 
> 
> How to eliminate it?

Enable iGMP Snooping to block it.

Web sfp - LAN - IGMP Snooping (Enable)

rapdodge, 2024-10-19:
> > > this could be IPTV ?
> > 
> > 
> > Maybe, but I didn't assign any VLAN to it...
> > ![{84F5DF55-14D9-41AB-83D4-0FE57998DCD0}](https://private-user-images.githubusercontent.com/33513626/378075479-76c93305-33d1-431f-98dc-b95c97d269ad.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjkzMzkzMDksIm5iZiI6MTcyOTMzOTAwOSwicGF0aCI6Ii8zMzUxMzYyNi8zNzgwNzU0NzktNzZjOTMzMDUtMzNkMS00MzFmLTk4ZGMtYjk1Yzk3ZDI2OWFkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDE5VDExNTY0OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQzYTIzMTU5MWRkMmU5MWE3Zjg1NGMxN2NlNzQ4YWNiZDVmY2U0ZTg5MzI0ZGI2MWRlZTZmMjEyYTljOGE5ZDEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.wDrzXxIhiorIHltLUQcVmZTtTgOjZLQBfdkU_yM9VoA)
> > How to eliminate it?
> 
> Enable iGMP Snooping to block it.
> 
> Web sfp - LAN - IGMP Snooping (Enable)

Thanks, it's slowing down right now

closed 2024-10-19 by rapdodge