Anime4000/RTL960x #330: Recommend stick for EPON network?

dlfrks, 2024-11-04:
I'm not sure which stick to use for EPON networks.
Can ODI DFP-34X-2C2 be used in EPON networks?

Anime4000, 2024-11-04:
should be possible, just need change the mode

giovannybm, 2024-11-06:
you can use xpon like DFP-34X-2C2, is compatible with xpon, epon

dlfrks, 2024-11-16:
I tried EPON network using DFP-34X-2C2 stick.

I used the following command:
```
flash set ELAN_MAC_ADDR <stock modem's mac address>
flash set MAC_KEY <calculated from mac address>
flash set OUI <first to sixth values ​​of the mac address>
flash set PON_MODE 2
```
After inserting the fiber it came online for a while and then the mac address was immediately blocked (returning to stock modem also does not work).

Should I get more information from my stock modem?
`OMCI_*` and `PON_VENDOR_ID` seem to be used only in GPON, but what other keys are used in EPON?

vtta, 2025-01-21:
> flash set ELAN_MAC_ADDR <stock modem's mac address>
> flash set MAC_KEY <calculated from mac address>

I am using the HSGQ version of the same stick on a EPON network, and I got it working by filling the MAC address with that of the original modem.
The `MAC_KEY` is calculated as usual with `echo -n "hsgq1.9aMACGOESHERE" | md5sum`.
I don't know whether this is strictly required, because after I flashed the IGD mode firmware and did a factory reset the MAC becomes something very weird and I have to change it.