Anime4000/RTL960x #335: Replacing HG8240H5 with DFP-34X-2C3 @ ER-X-SFP: no TX power

snizovtsev, 2024-11-10:
Hello, I'm trying to ditch my Huawei ONT with DFP-34X-2C3 (same as C2 but with green connector). It comes with V1.0-220923 factory firmware. My ISP is Mtel Montenegro (MTS Serbia branch).

My configuration looks like this:

[image]

In addition, I've tried setting `OMCI_OLT_MODE` from 0 to 1.

I've generated MAC KEY using with your `VsAuthKeyGen.exe` using my new HW version and MAC. Unfortunately, I've lost my original MAC KEY because it seems backup xml does not contain it.

Unfortunately, the stick reports that there are no TX power at all and even won't try to authenticate the connection. RX power looks similar to what I get with Huawei:

[image]
[image]
[image]

Could you help me with any suggestions related to this step?

Thank you in advance!

vuducdong, 2024-11-10:
you change mac sfp to ec84b444d3e6

and mac key to 1245734692fb7026c656cfd5ceb382b2

And fill any for loid and loid pass

Ex: 

Loid = 12345678
Loid pass = 123456

snizovtsev, 2024-11-10:
@vuducdong your MAC helped me, thank you very much! I got stable O5 state now:

[image]

So, It seems problem relies in `VsAuthKeyGen.exe`. It outputs the following for your key pair:

```
$ wine VsAuthKeyGen.exe EC84B444D3E6 168D.A
7E6B5902B10A57E8C685880F1708BC23 
```

How do I generate my own unique MAC KEY then?

snizovtsev, 2024-11-10:
I just figured out that `echo -n "hsgq1.9aEC84B444D3E6" | md5sum` algo from `providerforum.nl` matches `1245734692fb7026c656cfd5ceb382b2`. I will try to clone my original Huawei MAC in that way.

closed 2024-11-10 by snizovtsev

Anime4000, 2024-11-11:
> I just figured out that `echo -n "hsgq1.9aEC84B444D3E6" | md5sum` algo from `providerforum.nl` matches `1245734692fb7026c656cfd5ceb382b2`. I will try to clone my original Huawei MAC in that way.

I have link the Keygen in the big menu:
[image]

second, GPON doesn't care mac address unless you on EPON network