Anime4000/RTL960x #339: Vlan for Voice configuration for $200

qatar2030, 2024-11-26:
Hi
I replaced my Gpon Huawei router with ODI Realtek DFP-34X-2C2 and its work fine by creating Vlan10 for internet with PPPoE on sfp port.
am still suffering from configuration issue with Vlan30 for Voice or VoIP the scenario is:
[image]
this is address list, dhcp, etc for your vlan30
IP Acquisition Mode:DHCP
IP Address/Subnet Mask:xx.xxx.xxx.xxx/255.255.224.0
Gateway: xx.xxx.xxx.x
DNS Servers: xxx.xx.xxx.xx,xxx.xx.xxx.xx

if i have to hire someone to do it fro me online i have no problem for that I will pay him $200 via paypal
best regards

ToTheCLI, 2024-11-26:
Check your telegram, I have successfully accomplished this with STC SIP VOIP in KSA

qatar2030, 2024-12-02:
any help please

billbarni, 2024-12-08:
I have similar need...

I have a working DFP-34X-2C2 (and 2C3) for internet. But VOIP never authenticates.
I have it behind a OpnSense that grabs the VLANs and sets them for WAN, but the VOIP VLAN, despite working DHCP, does not authenticate.