Anime4000/RTL960x #344: hsgq z110 bandwidth limit to 30Mbps

DonuyThompson, 2024-12-09:
My ISP is Maxis 1000Mbps package.

I am usinghsgq z110. I flash the firmware to Z110_HSGQ_241129.tar
I also follow the instruction perform necessary setup stick. I am able to connect to the ISP via PPPoE.
However, when perform [speedtest](https://fast.com/). The download speed cap at 30Mbps and upload is below 1Mbps.
Plug the fiber cable back to original nokia G-1425G-B GPON, the speed is usual 960M/290M.
This is the Chipset of the GPON stick:

# cat /proc/cpuinfo
system type             : RTL8672
processor               : 0
cpu model               : 56322
BogoMIPS                : 299.00
tlb_entries             : 64
mips16 implemented      : yes
#

#  cat /etc/version
V1.0.5-241129 --  Fri 29 Nov 2024 11:25:01 AM CST
#

# diag
RTK.0> bandwidth get egress port all
port: 0  rate:4194296
         queue: 0  apr-index: 0
         queue: 1  apr-index: 0
         queue: 2  apr-index: 0
         queue: 3  apr-index: 0
         queue: 4  apr-index: 0
         queue: 5  apr-index: 0
         queue: 6  apr-index: 0
         queue: 7  apr-index: 0
port: 1  rate:4194296
         queue: 0  apr-index: 8
         queue: 1  apr-index: 8
         queue: 2  apr-index: 8
         queue: 3  apr-index: 8
         queue: 4  apr-index: 8
         queue: 5  apr-index: 8
         queue: 6  apr-index: 8
         queue: 7  apr-index: 8
port: 2  rate:4194296
         queue: 0  apr-index: 0
         queue: 1  apr-index: 0

--More--
         queue: 2  apr-index: 0
         queue: 3  apr-index: 0
         queue: 4  apr-index: 0
         queue: 5  apr-index: 0
         queue: 6  apr-index: 0
         queue: 7  apr-index: 0
RTK.0> andwidth get ingress port all
        ^Parse error
RTK.0> bandwidth get ingress port all
port: 0 rate:4194296
port: 1 rate:4194296
port: 2 rate:4194296
RTK.0> port get status port all
Port Status Speed    Duplex TX_FC RX_FC
---- ------ -----    ------ ----- -----
0    Up     2.5G     Full   Dis   Dis
1    Down   10M      Half   En    En
2    Down   1000M    Full   Dis   Dis
RTK.0>

Anime4000, 2024-12-10:
this ONU need set VLAN and connect without VLAN

vuducdong, 2024-12-10:
@DonuyThompson Try using SFU firmware

DonuyThompson, 2024-12-10:
> @DonuyThompson Try using SFU firmware

Thank you, if possible give me the appropriate update

vuducdong, 2024-12-10:
[Z110_HSGQ_SFU_240625_2.zip](https://github.com/user-attachments/files/18076787/Z110_HSGQ_SFU_240625_2.zip)

Please unzip this file to tar, then choose tar file to upgrade (up 2 times for 2 images)

Before, remember unplug fiber cable

After upgrade done, copy mac and mackey, use this command to reset factory

flash default hs
reboot

Finally, fill in all necessary parameters for setup onu (include mac and mackey above).

> > @DonuyThompson Try using SFU firmware
> 
> Thank you, if possible give me the appropriate update

DonuyThompson, 2024-12-10:
> [Z110_HSGQ_SFU_240625_2.zip](https://github.com/user-attachments/files/18076787/Z110_HSGQ_SFU_240625_2.zip)
>
> Please unzip this file to tar, then choose tar file to upgrade (up 2 times for 2 images) [...]

No connection

[image]

vuducdong, 2024-12-10:
What do you mean????

DonuyThompson, 2024-12-10:
> What do you mean????

[https://streamable.com/35kjqj](url)

vuducdong, 2024-12-10:
Please show mib 84, 171 and 329

omcicli mib get 84
omcicli mib get 171
omcicli mib get 329

@DonuyThompson

DonuyThompson, 2024-12-11:
> Please show mib 84, 171 and 329
> 
> omcicli mib get 84 omcicli mib get 171 omcicli mib get 329
> 
> @DonuyThompson

login: admin
Password:

BusyBox v1.22.1 (2024-06-25 17:47:09 CST) built-in shell (ash)
Enter 'help' for a list of built-in commands.

# omcicli mib get 84
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
# omcicli mib get 171
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
ExtVlanTagOperCfgData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
# omcicli mib get 329
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VEIP
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0x0601
AdminState: 1
OperState: 1
InterDomainName: 0x00000000000000000000000000000000000000000000000000
TcpUdpPtr: 0x0000
IansAssignPort: 0
CustomizedSlotId: 0xff
=================================
#

vuducdong, 2024-12-11:
@DonuyThompson 

According to the video link you sent above, I see that in the vlan section there are 9 and 114, while your mib 84 and 171 information is empty. Do you have fiber attached?

check OLT type:

omcicli mib get 131

After fill gpon sn, ploam pass, vendor id, hardware version, software version, Have you committed and rebooted yet?

Example:

flash set PON_VENDOR_ID ALCL
flash set GPON_SN ALCLxxxxxxx
flash set GPON_PLOAM_FORMAT 1
flash set GPON_PLOAM_PASSWD 31323334353637383930
flash set LOID 12345678
flash set LOID_PASSWD 123456
flash set OMCI_OLT_MODE 3
flash set OMCC_VER 128
flash set OMCI_FAKE_OK 1
flash set HW_HWVER 3FExxxxxxx
flash set HW_CWMP_PRODUCTCLASS G-1425G-B
flash set GPON_ONU_MODEL GPON,ONT
flash set HW_CWMP_MANUFACTURER NOKIA

nv setenv sw_custom_version0 3FExxxxxxx
nv setenv sw_custom_version1 3FExxxxxxx
reboot

DonuyThompson, 2024-12-11:
> Please show mib 84, 171 and 329
> 
> omcicli mib get 84 omcicli mib get 171 omcicli mib get 329
> 
> @DonuyThompson

This is the measurement with the fiber cable installed.

login: admin
Password:

BusyBox v1.22.1 (2024-06-25 17:47:09 CST) built-in shell (ash)
Enter 'help' for a list of built-in commands.

# omcicli mib get 84
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityID: 0xe102
FilterTbl[0]: PRI 0,CFI 0, VID 9
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0xe10b
FilterTbl[0]: PRI 0,CFI 0, VID 9
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0xff02
FilterTbl[0]: PRI 0,CFI 0, VID 114
FwdOp:  0x10
NumOfEntries: 1
=================================
# omcicli mib get 171
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
ExtVlanTagOperCfgData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0xe01
AssociationType: 10
ReceivedFrameVlanTagOperTableMaxSize: 0
InputTPID: 0x8100
OutputTPID: 0x8100
DsMode: 0
ReceivedFrameVlanTaggingOperTable
INDEX 0
Filter Outer   : PRI 8,VID 1000, TPID 5
Filter Inner   : PRI 8,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1
Treatment Inner   : PRI 9,VID 9, TPID 3
INDEX 1
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 8,VID 1000, TPID 5, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 1
Treatment Inner   : PRI 8,VID 9, TPID 2
INDEX 2
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 15,VID 0, TPID 5, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 0, TPID 2
INDEX 3
Filter Outer   : PRI 14,VID 4096, TPID 5
Filter Inner   : PRI 14,VID 4096, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 4096, TPID 3
INDEX 4
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 14,VID 4096, TPID 5, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 3
Treatment Inner   : PRI 15,VID 4096, TPID 2
AssociatedMePoint: 0x601
DscpToPbitMapping:
        0x000000
        0x249249
        0x492492
        0x6db6db
        0x924924
        0xb6db6d
        0xdb6db6
        0xffffff

Parsed DscpToPbitMapping:
dscp: 0 => pbit: 0
dscp: 1 => pbit: 0
dscp: 2 => pbit: 0
dscp: 3 => pbit: 0
dscp: 4 => pbit: 0
dscp: 5 => pbit: 0
dscp: 6 => pbit: 0
dscp: 7 => pbit: 0
dscp: 8 => pbit: 1
dscp: 9 => pbit: 1
dscp: 10 => pbit: 1
dscp: 11 => pbit: 1
dscp: 12 => pbit: 1
dscp: 13 => pbit: 1
dscp: 14 => pbit: 1
dscp: 15 => pbit: 1
dscp: 16 => pbit: 2
dscp: 17 => pbit: 2
dscp: 18 => pbit: 2
dscp: 19 => pbit: 2
dscp: 20 => pbit: 2
dscp: 21 => pbit: 2
dscp: 22 => pbit: 2
dscp: 23 => pbit: 2
dscp: 24 => pbit: 3
dscp: 25 => pbit: 3
dscp: 26 => pbit: 3
dscp: 27 => pbit: 3
dscp: 28 => pbit: 3
dscp: 29 => pbit: 3
dscp: 30 => pbit: 3
dscp: 31 => pbit: 3
dscp: 32 => pbit: 4
dscp: 33 => pbit: 4
dscp: 34 => pbit: 4
dscp: 35 => pbit: 4
dscp: 36 => pbit: 4
dscp: 37 => pbit: 4
dscp: 38 => pbit: 4
dscp: 39 => pbit: 4
dscp: 40 => pbit: 5
dscp: 41 => pbit: 5
dscp: 42 => pbit: 5
dscp: 43 => pbit: 5
dscp: 44 => pbit: 5
dscp: 45 => pbit: 5
dscp: 46 => pbit: 5
dscp: 47 => pbit: 5
dscp: 48 => pbit: 6
dscp: 49 => pbit: 6
dscp: 50 => pbit: 6
dscp: 51 => pbit: 6
dscp: 52 => pbit: 6
dscp: 53 => pbit: 6
dscp: 54 => pbit: 6
dscp: 55 => pbit: 6
dscp: 56 => pbit: 7
dscp: 57 => pbit: 7
dscp: 58 => pbit: 7
dscp: 59 => pbit: 7
dscp: 60 => pbit: 7
dscp: 61 => pbit: 7
dscp: 62 => pbit: 7
dscp: 63 => pbit: 7
=================================
=================================
EntityId: 0xf0f
AssociationType: 0
ReceivedFrameVlanTagOperTableMaxSize: 0
InputTPID: 0x8100
OutputTPID: 0x8100
DsMode: 0
ReceivedFrameVlanTaggingOperTable
INDEX 0
Filter Outer   : PRI 15,VID 0, TPID 0
Filter Inner   : PRI 15,VID 114, TPID 5, EthType 0x00
Treatment Outer   : PRI 15,VID 0, TPID 0, RemoveTags 0
Treatment Inner   : PRI 0,VID 114, TPID 2
AssociatedMePoint: 0xf0f
DscpToPbitMapping:
        0x000000
        0x249249
        0x492492
        0x6db6db
        0x924924
        0xb6db6d
        0xdb6db6
        0xffffff

Parsed DscpToPbitMapping:
dscp: 0 => pbit: 0
dscp: 1 => pbit: 0
dscp: 2 => pbit: 0
dscp: 3 => pbit: 0
dscp: 4 => pbit: 0
dscp: 5 => pbit: 0
dscp: 6 => pbit: 0
dscp: 7 => pbit: 0
dscp: 8 => pbit: 1
dscp: 9 => pbit: 1
dscp: 10 => pbit: 1
dscp: 11 => pbit: 1
dscp: 12 => pbit: 1
dscp: 13 => pbit: 1
dscp: 14 => pbit: 1
dscp: 15 => pbit: 1
dscp: 16 => pbit: 2
dscp: 17 => pbit: 2
dscp: 18 => pbit: 2
dscp: 19 => pbit: 2
dscp: 20 => pbit: 2
dscp: 21 => pbit: 2
dscp: 22 => pbit: 2
dscp: 23 => pbit: 2
dscp: 24 => pbit: 3
dscp: 25 => pbit: 3
dscp: 26 => pbit: 3
dscp: 27 => pbit: 3
dscp: 28 => pbit: 3
dscp: 29 => pbit: 3
dscp: 30 => pbit: 3
dscp: 31 => pbit: 3
dscp: 32 => pbit: 4
dscp: 33 => pbit: 4
dscp: 34 => pbit: 4
dscp: 35 => pbit: 4
dscp: 36 => pbit: 4
dscp: 37 => pbit: 4
dscp: 38 => pbit: 4
dscp: 39 => pbit: 4
dscp: 40 => pbit: 5
dscp: 41 => pbit: 5
dscp: 42 => pbit: 5
dscp: 43 => pbit: 5
dscp: 44 => pbit: 5
dscp: 45 => pbit: 5
dscp: 46 => pbit: 5
dscp: 47 => pbit: 5
dscp: 48 => pbit: 6
dscp: 49 => pbit: 6
dscp: 50 => pbit: 6
dscp: 51 => pbit: 6
dscp: 52 => pbit: 6
dscp: 53 => pbit: 6
dscp: 54 => pbit: 6
dscp: 55 => pbit: 6
dscp: 56 => pbit: 7
dscp: 57 => pbit: 7
dscp: 58 => pbit: 7
dscp: 59 => pbit: 7
dscp: 60 => pbit: 7
dscp: 61 => pbit: 7
dscp: 62 => pbit: 7
dscp: 63 => pbit: 7
=================================
# omcicli mib get 329
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VEIP
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0x0601
AdminState: 0
OperState: 0
InterDomainName: 0x00000000000000000000000000000000000000000000000000
TcpUdpPtr: 0x0000
IansAssignPort: 0
CustomizedSlotId: 0xff
=================================
#

vuducdong, 2024-12-11:
@DonuyThompson try with vlan 9 for internet (on ONT from ISP is vlan 1000, right?)

Tag vlan 9 on XPON ONU Z110 or change to auto then tag on WAN router

DonuyThompson, 2024-12-11:
Unfortunately there is no connection

kim12355, 2025-01-28:
@vuducdong i have one bought from MinhDang. I have just upgraded to firmware you mentioned above. Dial up pppoe is ok. But sometimes ping outside very high (ping to google.com about 8xx ms). I have to reboot hsgq z110 and it will normal. I was reset setting on webpage but not solve.

vuducdong, 2025-01-28:
> @vuducdong i have one bought from MinhDang. I have just upgraded to firmware you mentioned above. Dial up pppoe is ok. But sometimes ping outside very high (ping to google.com about 8xx ms). I have to reboot hsgq z110 and it will normal. I was reset setting on webpage but not solve. 
> 

Please check PON Status - RX Power, TX Power, and try to change LAN cable between onu and router.

kim12355, 2025-01-28:
> > I have checked but all normal. About cable it is ugreen cat7, i bought it 3 months ago
> > [image]
> > [image]
> > [image]
> > > > [@vuducdong](https://github.com/vuducdong) i have one bought from MinhDang. I have just upgraded to firmware you mentioned above. Dial up pppoe is ok. But sometimes ping outside very high (ping to google.com about 8xx ms). I have to reboot hsgq z110 and it will normal. I was reset setting on webpage but not solve.
> > > 
> > > 
> > > Please check PON Status - RX Power, TX Power, and try to change LAN cable between onu and router.
> 
> Fiber cable is SC/APC (green) or SC/UPC (blue) and check the same with port fiber on ONU.
> 
> Try to fill mac and mackey to mackey status change from fail to success.

It is green port. Where i can find mac key?

vuducdong, 2025-01-28:
> It is green port. Where i can find mac key?

You can use any mac address,and then apply to formula:

hsgq1.9a + MAC address (upper case all)

hsgq1.9aD49AA0453190

Finally, paste it to web to change md5hash.

Exam: https://www.md5hashgenerator.com

Result is md5hash is mackey

kim12355, 2025-01-30:
> You can use any mac address,and then apply to formula:
>
> hsgq1.9a + MAC address (upper case all)
>
> hsgq1.9aD49AA0453190
>
> Finally, paste it to web to change md5hash.
>
> Exam: [...]

I set priority port 2 to 1 and it havent happened for 2 days. Seem the problem is solved.