Anime4000/RTL960x #348: VSOL V2801F WebUI

mondychan, 2024-12-12:
hi, i have sucesfully connected via UART and TELNET to my VSOL V2801F gpon-on-stick module, 
backed up all the nessecary partitions and variables, 
and now i want to flash the custom Anime4000 firmware, 
yet my webui doesnt allow that, no option to upload firmware

i loged in using the  admin, stdONU101 combo

is there way to flash over telnet/tftp ?

thank you kindly for help

[image]
[image]

Anime4000, 2024-12-13:
may I know which firmware are you attempt to flash?

mondychan, 2024-12-13:
i was hoping for https://github.com/Anime4000/RTL960x/tree/main/Firmware/V2801F

possibly V2801F_V1.9.0-240614.tar

Anime4000, 2024-12-14:
probably not match hardware revision you have, the WebGUI just a GUI, under the hood remain same

Alzwei, 2026-01-16:
Use "http://192.168.1.1/upgrade.asp", defore reset default but dont reboot "flash_eraseall /dev/mtd3", if dont do this - you lost ip connection and change this maybe only in the UART.