Anime4000/RTL960x #367: Luleey LL-XS2510 analysis

totkeks, 2025-01-17:
Hey there,

thanks for this interesting repository.
I got the Luleey variant LL-XS2510.

Starting dumping the mtd partitions and then going through the files.

Most interesting are many of those files under `/lib`, because in their source they mention `license=GPL`, but I don't see the source nearby.
Did you ever try to ask Realtek for the sources, since that's clearly a requirement for GPL?

I was inspired by the `pon.wiki` project and their custom `was-110` firmware and wanted something similar for this stick.

The WebUI is some ASP scripts from 2003. It could really use an update. 😅

And maybe, using this project, fix the missing DDM chip so ethtool / i2c can show actual information.

Or activate the built-in SMNP functionality to query system stats from the router.
And if that doesn't work, create a JSON endpoint that returns system stats. CPU, Memory, Temperature, PON status, laser data and what not.

Currently trying to understand the boot process and write it all down.
There are tons of files that can just be deleted, because someone took the SDK from Realtek and just threw some stuff on it. And then someone else. And then another someone.

Besides the diagnostic tool `diag` and the webserver `boa` (with pages compiled in), the web pages use most of the space. And the majority of them aren't even used.

Anime4000, 2025-01-18:
Hi, Welcome!

# GPL
Yes, no source code is found, even license is GPL, we have tried to ask the source code, no replied from Realtek and HSGQ/ODI (who made Luleey SFP Stick) modified source code can be bought at cost of money

Thing is, no GCC compiler are understand RTL960B, RTL9601CI and RTL9601D used, it appear a modified MIPS Architecture ([Lexra LX5280](https://wikidevi.wi-cat.ru/Lexra_LX5280)), thus can't compile simple Hello World

# EEPROM
About DDM, you really need bought directly, HSGQ/ODI have release V08 Sticks last year which is has DDM
| Info          | V06 | V08 |
|---------------|-----|-----|
| DDM           | No  | Yes |
| Change EEPROM | Yes | No (Read Only) |
| TTL           | Yes | No |
| C2K           | No | Yes |

For me personally, I prefer V06, as you can modify SFP I2C EEPROM via [CH341a Programmer](https://www.hitoha.moe/read-sfp-i2c-via-ch341a-programmer/) which allow you to fix 2.5Gbps Information to make OpenWRT auto nego, see #250

V08 no longer have TTL, last I tested, it only have C2K communication, I think it replace TTL

# WebGUI
Yeah, the `boa` is default for Realtek Router/WiFi devices, even newer RTL9607DQ (ARM64) still use `boa` in 2024, max speed I tried is 1211Mbps `boa` can serve based on RTL9607DQ

# Limitation
RTL960x MIPS family very limited in resources, it only have 32MB of DDR2 build in RAM into SoC, even someone managed to reverse engineer whole thing and build into OpenWRT, RAM is limiting function, unlike Huawei MA5671a SoC has 64MB RAM and uses OpenWRT

due to WebGUI is like year 2003, I have managed to rebuild WebGUI to modern HTML5, and yes, I remove all *.asp file and replace with [HTML5 *.asp](https://github.com/Anime4000/RTL960x/tree/main/Firmware_Mod#webgui)

# Open GPON Initiative
[image]

I have planned to make GPON more open and compatible to many OLT, even someone in China managed to make RTL9601D ONT and use VSOL as the firmware

I no expert on Reverse Engineer, but I do what I can, and everyone here do the same, making patches, modify, etc... just to make SFP work with their OLT...

I am glad that you @totkeks learning boot process, if we could, reverse engineer the whole firmware binary into source code that we can compile automatically into firmware package

totkeks, 2025-01-18:
Thanks for the detailed reply! Is there any way to reach you via discord maybe?

I have seen the update on the Luleey webpage and thought I would buy that to have DDM, but now that I have access to the firmware, I thought about just adding some other trick to fix that.

> it appear a modified MIPS Architecture ([Lexra LX5280](https://wikidevi.wi-cat.ru/Lexra_LX5280)), thus can't compile simple Hello World

Just searching for LX5280 showed me multiple results with patches for the linux kernel and GCC. So that actually shouldn't be a problem.
https://gist.github.com/hackpascal/6d661db5ca3f4ed454c314363034a869

> it only have 32MB of DDR2 build in RAM

according to the specs on the Luleey webpage it is supposed to have 64MB. But I also only see around 32MB in the console. Though it is weird that I see one process with 80% memory and another with 40% memory, making that 120%. So there is some issue there.
The actual memory size would be shown in the serial console output of the bootloader, but I don't want to open my device to look at that.

> I have planned to make GPON more open and compatible to many OLT, even someone in China managed to make RTL9601D ONT and use VSOL as the firmware

It is a huge pity, that there is no open source GPON stack. This would be so amazing for the whole community / world. Because then we could start making custom firmware with up-to-date software versions.
But no, companies choose to do that behind closed doors.

> if we could, reverse engineer the whole firmware binary into source code that we can compile automatically into firmware package

I think that is illegal in most countries. We are not allowed to reverse engineer the code. One could do it locally on their machine, but you are not allowed to publish the results.

What I did for example is dump all the filesystems.
So, I now have a copy of the `boot` block, the `config` block mounted to `/var/config`, the two u-boot environments `env` and `env2`, and then the two firmware versions each with their `kernel0` and `rootfs0` and `kernel1` and `rootfs1`. 
These are versions `v1.0-220923` and `v1.0-220304`.

Here is the list of files I extracted. There is a lot of stuff that isn't actually needed.
Like the stick can do GPON and EPON. But it can also act as a router, so it has firewall functionality, VLAN and even WIFI configuration, both in the config files as well as the web UI.
Could probably reduce all that to the bare minimum as a starting point.

Compiling your own firmware will only be possible to a certain extend.
We would have to reuse the kernel and the kernel modules, as those are closed source.
But then it should be possible to compile additional libraries and binaries against this kernel and the existing libraries.
Especially using their libc, libgcc and lib-ld. Though I am just assuming, haven't done such things in the past.
The kernel is very old though. Version 2.6.30, so that makes it from the year 2009.

There is a download of compatible source code here https://www.boxmatrix.info/wiki/Property:RTL96
but I just looked, there seems to be no patched version of the GCC to build for that hardware. I guess that's not a requirement with GPL. 😂

<details><summary>Files in the root filesystem</summary>
<pre>
/.lstripped
/bin
/bin/ShowStatus
/bin/arp
/bin/ash
/bin/awk
/bin/boa
/bin/brctl
/bin/busybox
/bin/cat
/bin/checkomci
/bin/chmod
/bin/cksum
/bin/cmp
/bin/configd
/bin/consoleset
/bin/cp
/bin/cut
/bin/date
/bin/dbg_tool
/bin/df
/bin/diag
/bin/diff
/bin/dropbear
/bin/ebtables
/bin/echo
/bin/egrep
/bin/eponoamd
[... 857 lines cut ...]
/var
</pre>
</details>

I'm currently trying to extract the kernel from the uimage, but for some reason that doesn't seem to work. I'm getting some nonsense there and then neither `binwalk` nor `nm` or `objdump` can extract the actual kernel.
Because I wanted to check which modules are backed in there.

```
binwalk kernel0

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
165351        0x285E7         ESP Image (ESP32): segment count: 2, flash mode: QUIO, flash speed: 40MHz, flash size: 4MB, entry address: 0xa1000821, hash: none
806923        0xC500B         ESP Image (ESP32-S2): segment count: 2, flash mode: QUIO, flash size: 4MB, entry address: 0x228e21, hash: none
836719        0xCC46F         ESP Image (ESP32): segment count: 2, flash mode: QUIO, flash speed: 40MHz, flash size: 2MB, entry address: 0x2e030c21, hash: none
1135363       0x115303        ESP Image segment count: 12, flash mode: FAST_READ, flash size: 4MB, entry address: 0xb1af5c, hash: none
1215583       0x128C5F        ESP Image (ESP32-C6): segment count: 12, flash mode: FAST_READ, flash size: 4MB, entry address: 0xa627d7, hash: none
2383904       0x246020        Linux kernel version 2.6.30
2404032       0x24AEC0        CRC32 polynomial table, little endian
2555319       0x26FDB7        Neighborly text, "neighbor %.2x%.2x.%.2x:%.2x:%.2x:%.2x:%.2x:%.2x lost on port %d(%s)(%s)"
2637615       0x283F2F        Neighborly text, "NeighborTable_gettingTable_get"
2637645       0x283F4D        Neighborly text, "NeighborTable_setingTable_set"
```

Other than that no current plans. Maybe document the boot process and upload that to a git repository. The idea is to describe the flow on boot and also describe the purpose of each of those files, scripts and such, to make sense of the whole firmware.

Anime4000, 2025-01-19:
> Luleey webpage it is supposed to have 64MB

So, we has been lied by bootloader? 🤣 This it's possible to  make OpenWRT based? or modify heavily what ever is base on...

> There is a download of compatible source code here

Well, if you look RTL9607DQ, is a ARM64 (Cortex A55), whole RTL96x family is PON Router, so far My ISP supply D-Link DPN-FX3060V that use RTL9607DQ also having issue with some OLT, especially Fiberhome OLT which having speed issue, then My ISP changed to another vendor Skyworth GN360V that use MediaTek EcoNet and solve the issue what Realtek is facing.

What I guess, Realtek Implementation of PON are not matured as MediaTek, this what limit as user and ISP who use Universal ONU Deployment which not all ONU can support on their FTTH network.

I guess, Realtek could make Open Source the firmware part, at least community can patch and test various of OLT, but this could give an issue to user try on Production network which can blackout whole neighbors internet...

> I just looked, there seems to be no patched version of the GCC to build for that hardware. I guess that's not a requirement with GPL.

Aww that's sad, they get away with it 😅

> Like the stick can do GPON and EPON. But it can also act as a router, so it has firewall functionality, VLAN and even WIFI configuration, both in the config files as well as the web UI.
Could probably reduce all that to the bare minimum as a starting point.

I and most everyone here afraid to modify heavily without killing the stick, thus, we just use what is possible, at least remove unrelated `*.asp` to reduce space

One more thing, if, Realtek release RTL9601E and use SONET OC-48 along side HiSGMII and 2500Base-X, this help border compatibility

totkeks, 2025-01-19:
Is there any way to easily contact you or discuss this debugging with other people that have commented and worked on your repo to collect all the data?

There isn't really much going on in that system.

We got the `omcid` and `configd` that manage the system configuration and communication with the ISP.

And before that some firmwares are loaded into the appropriate hardware.
One for the laser driver `rtkbosa` and another two with `europacli`.

Do you have any idea what `europa` that could be? Internet search isn't really helpful for niche information. Is that the codename of the SoC?

And then depending on EPON, GPON or AON, the `pondetect` is used. Or traffic is just passed through.

And there is dedicated hardware for that, which explains why the cpu is always displayed at 0% in the web UI.

I think there is a low chance of bricking things if keep the original services running.
Could even optimize the process a bit. Some of the scripts do basically nothing or literally nothing, because someone commented out all the lines. Or changed an `if` statement to always be `false`.