Anime4000/RTL960x #379: Pon stick Web UI is not accessible without fiber patch cord, but ...

CpServiceSpb, 2025-02-24:
There is no Web UI access of DFP-34X xpon stick without connected fiber patch cord.
Stick was plugged to Mikrotik, Keetic routers, some switches.

But there is Web UI access ...
Stick was plugged to Dlink DGS-3000 serie switch.

Why so ?

No any special settings were made on the stick.

Anime4000, 2025-02-25:
for Mikrotik, you need use ROS 7.17+ and do this:
✅ Ignore Rx LOS

because ODI act like normal Fiber Transceiver. Perhaps get an 2.5G SFP Media Converter to config the stick

CpServiceSpb, 2025-03-05:
> for Mikrotik, you need use ROS 7.17+ and do this: ✅ Ignore Rx LOS
> 
> because ODI act like normal Fiber Transceiver. Perhaps get an 2.5G SFP Media Converter to config the stick

I tried so - nothing changed.
And answering for my question:
Only some switches where it works without patchcord connected which I saw by myself are: DGS-3000, GSD-1020S.

Maybe because of LFP functionality lacking.

And as I heard not all but some mediaconverters, but I don' t know exactly which ones.