Anime4000/RTL960x #380: ODi DFP-34G-2C2 brick

anilkurt11, 2025-03-01:
ODI DFP-34G-2C2 cihazıma firmware yükledikten sonra erişim sağlayamıyorum. Ayrıca Nokia G-010G-R ont cihazına TTL ile bağlandım fakat Login incorrect hatası alıyorum. Yardımcı olurmusunuz ?

[image]
[image]

oneevil, 2025-03-12:
How did you get to the password entry window? Mine doesn't respond to anything and loads the kernel.

anilkurt11, 2025-03-12:
> How did you get to the password entry window? Mine doesn't respond to anything and loads the kernel.

Baudeaea 9600

oneevil, 2025-03-12:
This is the original firmware from the factory.

oneevil, 2025-03-12:
> Baudrate 9600

No power to it?

anilkurt11, 2025-03-13:
I connect the power. But the baudrate 115200 connection only sends data. It does not receive commands. Then, when I connect it as baudrate 9600, it receives commands but I get a login incorrect error. ONTUSER and S/N passwords do not match.

anilkurt11, 2025-03-13:
> > Baudrate 9600
> 
> No power to it?

I connect the power. But the baudrate 115200 connection only sends data. It does not receive commands. Then, when I connect it as baudrate 9600, it receives commands but I get a login incorrect error. ONTUSER and S/N passwords do not match.

anilkurt11, 2025-03-13:
@Anime4000  CIG Version: 3.09.08

U-Boot 2011.12.NA-svn145270 (Aug 18 2021 - 15:54:14)

Board: RTL9601D, CPU:112MHz, LX:200MHx, MEM:325MHz, Type:DDR2
DRAM: 32 MB
SPI-F: WINBOND/EF4018/MMIO16-1/ModeC 1x16 MB (plr_flash_info @ 81c3bcac)
*** Warning - bad CRC, using default environment

--major = f2, minor = 1
--major = f2, minor = 1
*** !!! Check Result: EEPA and EEPB are all valid
Device care eqvid: 3FE49165BCAA, Init serial
Net:   LUNA GMAC
Warning: eth device name has a space!

swcore_init
**************************************
*                                    *
*  KEY -- Enter console terminal     *
*                                    *
**************************************
waiting for your select
do_jffs2_fsload offset 0x80400000 loadaddr=0xb4088000..
get loadbit 0,bootbit 0
Get current Image = ImageA..

--- jffs2_part_info: partition number 2 for device sflash0 (sflash.0)

Loading 'uImage' from  CRAMFS Partition 'imagea' to 0x80400000.

Root Filesystem crc check successfully!

PRODUCT:  sfu
SS  :  noss
DSP :  nodsp
WIFI:  nowifi
XDSL:  nodsl
Software version:  3FE49717AOCK55
## Booting kernel from Legacy Image at 80400000 ...
   Image Name:   Linux Kernel Image
   Created:      2023-07-27   9:53:44 UTC
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    775867 Bytes = 757.7 KB
   Load Address: 80000000
   Entry Point:  80000000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK

Starting kernel ...

When I connect to Nokia G-010G-R model ONT device with 115200 baudrate and reset it, I can get this information but I can't send commands. The baudrate I send commands to is 9600. But I get a login incorrect error. Can you help?

oneevil, 2025-03-13:
@anilkurt11 Did you generate the password here https://hack-gpon.org/ont-nokia-g-010g-r/ ?

oneevil, 2025-03-13:
And what information do you need from it? Or do you want to program it to your needs?

anilkurt11, 2025-03-13:
> [@anilkurt11](https://github.com/anilkurt11)https://hack-gpon.org/ont-nokia-g-010g-r/ adresindeki şifreyi siz mi oluşturdunuz ?

Yes, I tried, but it didn't work.

anilkurt11, 2025-03-13:
> Ve ondan hangi bilgilere ihtiyacınız var? Yoksa ihtiyaçlarınıza göre programlamak mı istiyorsunuz?

I need information like mib show 256 omci version

oneevil, 2025-03-13:
> > Ve ondan hangi bilgilere ihtiyacınız var? Yoksa ihtiyaçlarınıza göre programlamak mı istiyorsunuz?
> 
> I need information like mib show 256 omci version

It's the same as on the web http://192.168.100.2/ -> More info

anilkurt11, 2025-03-13:
> > > Ve ondan hangi bilgilere ihtiyacınız var? Yoksa ihtiyaçlarınıza göre programlamak mı istiyorsunuz?
> > 
> > 
> > I need information like mib show 256 omci version
> 
> It's the same as on the web http://192.168.100.1/ -> More info
> 
> TurkTelekom? Turkcell? Superonline?

Turknet service provider connected to Turk Telekom infrastructure

anilkurt11, 2025-03-13:
> Turknet service provider connected to Turk Telekom infrastructure

The information on the 192.168.100.1 interface is limited. It does not allow connection. I get an o5 wrong error.

oneevil, 2025-03-13:
Use this https://hack-gpon.org/ont-huawei-ma5671a-fs-mod/

This works perfectly well for TurkTelecom.
The information on your local forums is not quite correct.

anilkurt11, 2025-03-13:
Yes. ont Id Default

oneevil, 2025-03-13:
> Yes. ont Id Default

fw_setenv onu_serial "Serial_number"
fw_setenv onu_ploam "DEFAULT"
fw_setenv omci_vendor_id "ALCL"
fw_setenv omci_equip_id "Serial_number"
fw_setenv omci_hw_ver "Hardware_version"
fw_setenv image0_version "Active_software_version"
fw_setenv image1_version "Standby_software_version"
fw_setenv sfp_vendor_name "ALACTELLUCENT"
fw_setenv sfp_part_name "G010SP"
fw_setenv sfp_vendor_rev "10"
fw_setenv sfp_part_serial "Serial_number"
fw_setenv sfp_date_code "230916"
uci set network.lct.macaddr=MAC_from_label+3
uci set network.host.macaddr=MAC_from_label
uci commit network
reboot

anilkurt11, 2025-03-13:
> Use this https://hack-gpon.org/ont-huawei-ma5671a-fs-mod/
> 
> This works perfectly well for TurkTelekom. The information on your local forums is not quite correct. ONT ID Check with HEX to ACSII converter all characters are readable?
I use  ODI DFP-34X-2C2

oneevil, 2025-03-13:
> > Use this https://hack-gpon.org/ont-huawei-ma5671a-fs-mod/
> > This works perfectly well for TurkTelekom. The information on your local forums is not quite correct. ONT ID Check with HEX to ACSII converter all characters are readable?
> > I use  ODI DFP-34X-2C2

Shouldn't be a problem either.
Show me what you're filling out.

KYTECHNGAMING, 2025-03-22:
Burda mıydınız arkadaşlar :)

usmcsas, 2025-03-24:
Buradayız 😃 Arkadaşlar merhaba aynı ONT bende de var huawei sfp mevcut. Fake O5 alıyorum.

oneevil, 2025-03-24:
> Can I install this mod directly from the SFP openWRT UI or need a flash tool or sth.

You can flash it via SSH

usmcsas, 2025-03-24:
> > Can I install this mod directly from the SFP openWRT UI or need a flash tool or sth.
> 
> You can flash it via SSH
> 
> Get current run image: fw_printenv committed_image
> 
> Send firmware bin to sfp: scp -O fs.bin root@192.168.1.10:/tmp/
> 
> Flash image: mtd -e image1 write /tmp/fs.bin image1 fw_setenv committed_image 1 fw_setenv image1_is_valid 1 reboot
> 
> After reboot: firstboot

Thank you but the file format is .img in that https://hack-gpon.org/ont-huawei-ma5671a-fs-mod/ is there anything wrong ?

oneevil, 2025-03-24:
Just rename filename to fs.bin
It doesn't make any difference what it's called

Troll338cz, 2026-09-27:
Can someone with this device try generate login with this code? 

Nokia G-010G-R might have non-standard password generator.

I extracted the code from [this](https://github.com/hack-gpon/hack-gpon.github.io/issues/444) fash dump.

```py
import hashlib
# from VOS_CfgParamGetByName("EepEqSerialNumber", a1, 16)
GPON_SN = "ACLCa1b2c3d4" # Must be 4 uppercase + 8 lowercase
text = GPON_SN + "-ONTUSER"
# from /usr/lib/libvos.so.0.0.0
charset = "ACDEFGHJKLMNPQRSTUVWXYZ2345679abcdefghijkmnpqrstuvwxyz"  # Base58
rawdigest = hashlib.sha256(text.encode("utf-8")).digest()
output = []
#  From /bin/Console
#  sprintf(v2, "%s-%s", v1, "ONTUSER");
#  hmac_sha256_pwd(v2, &cli_password, 16);
for i in range(16):
   byte_val = rawdigest[i]
   char_index = byte_val % 0x36
   output.append(charset[char_index])

print("".join(output))
```

pantigon, 2026-10-25:
> 29\. Can someone with this device try generate login with this code?
>      Nokia G-010G-R might have non-standard password generator.
>      I extracted the code from [this](https://github.com/hack-gpon/hack-gpon.github.io/issues/444) fash dump.
>      import hashlib
>      # from VOS_CfgParamGetByName("EepEqSerialNumber", a1, 16)
>      GPON_SN = "ACLCa1b2c3d4" # Must be 4 uppercase + 8 lowercase
>      text = GPON_SN + "-ONTUSER"
>      # from /usr/lib/libvos.so.0.0.0
>      charset = "ACDEFGHJKLMNPQRSTUVWXYZ2345679abcdefghijkmnpqrstuvwxyz"  # Base58
>      rawdigest = hashlib.sha256(text.encode("utf-8")).digest()
>      output = []
>      #  From /bin/Console
>      #  sprintf(v2, "%s-%s", v1, "ONTUSER");
>      #  hmac_sha256_pwd(v2, &cli_password, 16);
>      for i in range(16):
>         byte_val = rawdigest[i]
>         char_index = byte_val % 0x36
>         output.append(charset[char_index])
>      
>      print("".join(output))

`CIG Version: 3.09.08

U-Boot 2011.12.NA-svn145270 (Aug 18 2021 - 15:54:14)

Board: RTL9601D, CPU:112MHz, LX:200MHx, MEM:325MHz, Type:DDR2
DRAM: 32 MB
SPI-F: 204018/MMIO16-1/ModeC 1x16 MB (plr_flash_info @ 81c3bcac)
*** Warning - bad CRC, using default environment

--major = f2, minor = 1
--major = f2, minor = 1
*** !!! Check Result: EEPA and EEPB are all valid
Device care eqvid: 3FE49165BCAA, Init serial
Net:   LUNA GMAC
Warning: eth device name has a space!

swcore_init
**************************************
*                                    *
*  KEY -- Enter console terminal     *
*                                    *
**************************************
waiting for your select
do_jffs2_fsload offset 0x80400000 loadaddr=0xb4088000..
get loadbit 0,bootbit 0
Get current Image = ImageA..

--- jffs2_part_info: partition number 2 for device sflash0 (sflash.0)

Loading 'uImage' from  CRAMFS Partition 'imagea' to 0x80400000.

Root Filesystem crc check successfully!

PRODUCT:  sfu
SS  :  noss
DSP :  nodsp
WIFI:  nowifi
XDSL:  nodsl
Software version:  3FE49717AOCK55
## Booting kernel from Legacy Image at 80400000 ...
   Image Name:   Linux Kernel Image
   Created:      2023-07-27   9:53:44 UTC
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    775867 Bytes = 757.7 KB
   Load Address: 80000000
   Entry Point:  80000000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK

Starting kernel ...
`

[image]

I try it and wrong password.

Troll338cz, 2026-10-25:
Thank you, it was worth a try anyway, it's possibly wrong decompile or other ISP firmware. If you dump CramFS from U-Boot i can have a look at yours later.

pantigon, 2026-10-25:
> 31\. Thank you, it was worth a try anyway, it's possibly wrong decompile or other ISP firmware. If you dump CramFS from U-Boot i can have a look at yours later.

I re-flash [This](https://github.com/hack-gpon/hack-gpon.github.io/issues/444)

Troll338cz, 2026-10-25:
Then will look at the firmware again why it failed, also worth a try sending [CIG backdoor enable](https://github.com/YuukiJapanTech/CA8271x/blob/main/doc/rootShell.md#nokia-xs-010x-r) i haven't looked on how its implented on this device but might work.

pantigon, 2026-10-26:
> 31\. Then will look at the firmware again why it failed, also worth a try sending [CIG backdoor enable](https://github.com/YuukiJapanTech/CA8271x/blob/main/doc/rootShell.md#nokia-xs-010x-r) i haven't looked on how its implented on this device but might work.

It working. Thanks.

Troll338cz, 2026-10-26:
Can you please share the details of what happened after you send it, i will properly document the process for hack-gpon

pantigon, 2026-10-26:
> 32\. Can you please share the details of what happened after you send it, i will properly document the process for hack-gpon

[image]

Run script (don't show anything) then telnet as document.