Anime4000/RTL960x #382: Get DFP-34X-2C2 working at 2.5Gbps link speeds in Ubiquiti UCG-Fiber ?

d5aqoep, 2025-03-12:
I have 2 sticks from ODI DFP-34X-2C2 and it's newer HSGQ rebranded version. Whenever I insert them in my Ubiquiti UCG-Fiber that has 2 10G SFP ports, it connects only at 1G speeds. My ISP provides upto 2 Gbps bandwidth so it seems like I am getting restricted. The Ubiquiti interface allows only 1G and 10G speeds for these sticks. So is it possible to get the sticks to connect at their rated 2.5G speeds?

[img]https://i.imgur.com/PjD4R6M.jpeg[/img]

anilkurt11, 2025-03-12:
There are 1G-2.5G Lan options in the firmware found here https://github.com/zycsnd/ODI-DFP-34X-2C2. You can try it if you want.

d5aqoep, 2025-03-12:
What LAN_SDS_MODE needs to be set for it?

Anime4000, 2025-03-12:
if firmware 22xxxx, use
LAN SDS MODE 6 (2500 BaseX)

if firmware 24xxxx, use
LAN SPEED MODE 4 (HiSGMII + BaseX)

d5aqoep, 2025-03-13:
@Anime4000 Tried both modes and for both modes my UCG-Fiber syncs the SFP stick at 1G.

Anime4000, 2025-03-22:
have you tried contact Ubiquiti regarding this GPON Stick issue?

belkone, 2025-03-22:
no, it is based on what I found in myself and what Unifi GPT answered me on their website:

[image]

I have UCG-Fiber and ODI DFP-34X-2C3 and I will want to check it, but I am waiting for the media converter to arrive so I can go back to 1GBE in case of incompatibility :)