Anime4000/RTL960x #384: Stick firmware TR-069 functionality

CpServiceSpb, 2025-03-27:
Can somebody who has DFP-34X-.... firmware sources add TR-069 functionality to HGU (hybrid) firmware ?
That a stock could pass activation procedure for an ISP where TR-069 is used.

If so I can assist by some semi orking firmware (but no sources), stick with necessary hardware and testing environment.

CpServiceSpb, 2025-06-21:
Anime4000 can you contact me to discuss this question at Tellegram ?
I wrote you before to.

Anime4000, 2025-06-22:
in order TR069 to work, VLAN for TR069 must created via ME 84 and/or ME 171
if no VLAN for TR069, how suppose to connect to ISP Management

let say for argument sake

add TR069, then ISP push configuration which PON Stick don't have and don't know, still get rejected as each ISP TR069 different than each other

for example, my ISP has own TR069 and it's optional

that what I discover within my ISP network