Anime4000/RTL960x #387: Help with Vlans Configuration

AntoniooDidier, 2025-04-06:
Hello, thank you for sharing information here on GitHub, this thread has been very helpful. However, I need some assistance if possible. I am trying to replace my ISP router (Altice MEO) with the ODI DFP-34X-2C3, but I am having some difficulties. I cloned all the information from the original router to the SFP, including the PLOAM password, which is 10 spaces in ASCII or 20x20 in HEX. I can authenticate with O5 and receive the VLANs, but two things are happening:

I configure VLAN 840 on my Unifi (also clone -> MAC address of the original router 68:AA:C4:XX:XX:XX, DHCP option 60 "meods00", DHCP option 61 "68:AA:C4:XX:XX:XX"), I get an IP for a few days and then I get O7.

Sometimes, I cannot even get an IP on VLAN 840.

**My firmware version is the "M110_sfp_ODI_240408_rel240513+hwtc-fix"**

I will leave the configurations I was able to obtain from the original router:

`----------------------------------------------
WAN Info
----------------------------------------------
Interface:                     erouter0
MAC Address :                  68:AA:C4:XX:XX:XX
Description:                   Interface_erouter0
Type:                          IPoE
Vlan 802.1p:                   0
Vlan Mux ID:                   12
Vlan TPID:                     0x8100
IPv6:                          Enabled
IGMP Proxy:                    Enabled
IGMP Source:                   Enabled
MLD Proxy:                     Disabled
MLD Source:                    Disabled
NAT:                           Enabled
NAT Type:                      Masquerade
Firewall IPv4:                 Enabled
Firewall IPv6:                 Enabled
Force Disable Firewall IPv4:   Disabled
Force Disable Firewall IPv6:   Disabled
ARPPing:                       Enabled
ARPPing timeout(secs):         900
ARPPing number of repetitions: 3
Status:                        Connected
IPv4 address:                  85.244.xx.xx
IPv6 address:                  2001:8a0:61bb:aa00:6aaa:xxxx:xxxx:xxxx
Enable/Disable:                Enabled
Addressing Type IPv4:          DHCP
------DHCPv4------
Option 60 Vendor ID:           meods00
Option 61 IAID:
Option 61 DUID:
Subnet Mask:                   255.255.255.0
Default Gateway:               85.244.xx.xx
Option 125:
DNS Servers:                   212.55.154.174,212.55.154.190
Remaining Lease Time(secs):    3040
------DHCPv6------
Type:                          PrefixDelegation
Prefix:                        2001:8a0:61bb:aa00::/56
Preferred lifetime:            90000
Valid lifetime:                90000
Default gateway:
DNS Servers:                   2001:8a0:2106:5:212:55:154:174,2001:8a0:2206:5:212:55:154:190
----------------------------------------------`

And here are my MIBs:

`# omcicli mib get 131
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
OltG
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0x00
OltVendorId: 0x5054494e
EquipId:
Version: PQ
ToDInfo:
        Sequence number of GEM superframe: 0x0
        Timestamp: secs 0, nanosecs 0
=================================

# omcicli mib get 84
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityID: 0x0a
FilterTbl[0]: PRI 0,CFI 0, VID 832
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0x22
FilterTbl[0]: PRI 0,CFI 0, VID 840
FwdOp:  0x10
NumOfEntries: 1
=================================
=================================
EntityID: 0xffff
FilterTbl[0]: PRI 0,CFI 0, VID 12
FwdOp:  0x10
NumOfEntries: 1
=================================

# omcicli mib get 171
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
ExtVlanTagOperCfgData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0x03
AssociationType: 2
ReceivedFrameVlanTagOperTableMaxSize: 0
InputTPID: 0x8100
OutputTPID: 0x8100
DsMode: 0
ReceivedFrameVlanTaggingOperTable
INDEX 0
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 8,VID 12, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 4097, TPID 1, RemoveTags 1
Treatment Inner   : PRI 8,VID 840, TPID 4
INDEX 1
Filter Outer   : PRI 15,VID 4096, TPID 0
Filter Inner   : PRI 8,VID 10, TPID 0, EthType 0x00
Treatment Outer   : PRI 15,VID 4097, TPID 1, RemoveTags 1
Treatment Inner   : PRI 8,VID 832, TPID 4
AssociatedMePoint: 0x101
DscpToPbitMapping:
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
=================================
=================================
EntityId: 0x06
AssociationType: 10
ReceivedFrameVlanTagOperTableMaxSize: 0
InputTPID: 0x00
OutputTPID: 0x00
DsMode: 0
ReceivedFrameVlanTaggingOperTable
AssociatedMePoint: 0x601
DscpToPbitMapping:
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
=================================
=================================
EntityId: 0x08
AssociationType: 0
ReceivedFrameVlanTagOperTableMaxSize: 0
InputTPID: 0x00
OutputTPID: 0x00
DsMode: 0
ReceivedFrameVlanTaggingOperTable
AssociatedMePoint: 0x07
DscpToPbitMapping:
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
        0x000000
=================================
#`

I hope someone can help me... thank you very much!!!!

ToTheCLI, 2025-04-08:
Try older firmware

[M110_sfp_ODI_220923.tar](https://github.com/Anime4000/RTL960x/blob/main/Firmware/DFP-34X-2C2/M110_sfp_ODI_220923.tar)

AntoniooDidier, 2025-04-08:
> Try older firmware
> 
> [M110_sfp_ODI_220923.tar](https://github.com/Anime4000/RTL960x/blob/main/Firmware/DFP-34X-2C2/M110_sfp_ODI_220923.tar)

That was the first firmware that I have used, same problem...

> Maybe you need some TR069 reporting? I seen some issues here that some ISPs has requirement in having TR069 for their devices.

Maybe... What I can do to enable TR069 report? if that is possible 😬

SkateWarp, 2025-04-08:
First of all...

Do you suspect any of the VLANs that you receive is the one for TR-069? 
Can you view to the TR-069 config in the original ISP device? 
Also, I don't know much about TR-069 in general, aside it gives the ISP a way to control to the device or use it as some sort of telemetry protocol or both. 
As for the ODI, don't know much if is working. 
There's a discussion thread, but it doesn't say much:
https://github.com/Anime4000/RTL960x/discussions/366

MrSuicideParrot, 2025-04-09:
Hi!

 I have worked in the past on devices  of MEO ISP, and from my understanding when they use dedicated ONTs,  they don't have any IP configured on the ont. The only thing that has tr-69 is the router itself and doesn't use any dedicated vlan for it, they use the default internet vlan ( vlan 12) and the public IP of the router for managing via tr-69.

Regarding the problem, at least in the past (2 years ago) when I tried using an ODI in this isp, in the case of being an Altice OLT, it would give O5, but I wouldn't have internet due to a misconfiguration in the VLAN table.I would saw with Wireshark the downlink traffic in a VLAN where is not expected (VLAN 500) but I wouldn't have uplink.

I the end I  have never was able to configure an ODI ont in a reliable way in this ISPs every time the OLT was "PT Altice".

MrSuicideParrot, 2025-04-09:
Do you know which OLT is serving your house? There are Huawei and Altice olts.

AntoniooDidier, 2025-04-09:
> Do you know which OLT is serving your house? There are Huawei and Altice olts.

Yes i have that infromation on the first topic:

```plaintext
# omcicli mib get 131
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
OltG
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityId: 0x00
OltVendorId: **0x5054494e**
EquipId:
Version: PQ
ToDInfo:
        Sequence number of GEM superframe: 0x0
        Timestamp: secs 0, nanosecs 0
=================================
```

Its a altice PTIN :)

MrSuicideParrot, 2025-04-09:
Sorry, for not noticing that it was a PTIN OLT, when I tried I never had success with them...

Only three questions:
Do you see any pattern in the time it takes to loose the IP address in your router?
After having O7, if you reboot the ODI, does it work again?
Meanwhile did you notice any problem after connecting again the ISP ont (after having O7, in the ODI) ?

AntoniooDidier, 2025-04-09:
>Do you see any pattern in the time it takes to loose the IP address in your router?
Nope
>After having O7, if you reboot the ODI, does it work again?
>Meanwhile did you notice any problem after connecting again the ISP ont (after having O7, in the ODI) ?
I need contact the ISP to unlock my conn with OLT, even with original router i cant connect again.

closed 2025-04-22 by AntoniooDidier