Anime4000/RTL960x #389: ONU State O5 - Initial Status

danielkot, 2025-04-13:
Hello i have a problem with ONT LXT-010H-D.
Cloned from ZTE zxhn f660: serial number, PON Serial number, Hardware Version, Software Version, set fake_ok
Got ONU ID and VLan  as in screenshots
What's I'm doing wrong?

[image]

[image]

```
# omcicli mib get 84
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityID: 0x101
FilterTbl[0]: PRI 1,CFI 0, VID 1108
FwdOp:  0x04
NumOfEntries: 1
=================================
=================================
EntityID: 0x4101
FilterTbl[0]: PRI 1,CFI 0, VID 1108
FwdOp:  0x0c
NumOfEntries: 1
=================================
#
#
# omcicli mib get 171
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
ExtVlanTagOperCfgData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
#
```

Anime4000, 2025-04-13:
ME 171 is missing, for this, you need to use HGU Firmware instead to simulate ONT-Router

danielkot, 2025-04-14:
Where I can find this firmware?

danielkot, 2025-06-02:
The latest version fo me is V3.3.2L10. Sometimes I'll look to upgrade my LXT-010H-D.
PS: You also have leox ONT LXT-010H-D?