Anime4000/RTL960x #390: ODI DFP-DFP-34X-2C2: (O5 Status, Fleeting IP )

yoyo-youyou, 2025-04-13:
The ODI DFP-DFP-34X-2C2 modem had been working perfectly for several years until the ISP reported it needed repairs. After being offline for an entire morning, the modem stopped functioning properly. It briefly acquired an IP address, but it quickly cycled through several IPs (e.g., A -> B -> C->D ) before ultimately losing its IP completely.

Interestingly, when I connected an old modem that hadn’t been used for years, the connection returned to normal (and the IP did not cycle repeatedly). Upon reconnecting the original modem, I noticed that the password and GPON SN appeared unchanged, but the modem had downloaded new files with logs showing the following:

[Notice] Start to load a packet file by OMCI
[Notice] Finish to load a packet file!!
[Notice] Load Success!
[Notice] [OLT]: reset ont

How can I access the OMCI on a Huawei modem? This issue is preventing me from connecting to the internet (unable to obtain an IP 
 address). Thank you all for your help—it's greatly appreciated!
[image]

[image]

Anime4000, 2025-04-13:
show me `ME 84` and `171`

yoyo-youyou, 2025-04-14:
> show me `ME 84` and `171`

# omcicli mib get 84
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityID: 0x01
FilterTbl[0]: PRI 0,CFI 0, VID 4011
FwdOp:  0x04
NumOfEntries: 1
=================================
# omcicli mib get 171
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
ExtVlanTagOperCfgData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX

Thank you for helping to solve the problem.