Anime4000/RTL960x #391: SFP module causes reboot loop when fiber is connected during power-up

AntoniooDidier, 2025-04-13:
Hi, I'm experiencing a problem with the RTL960x firmware. When there's a power outage and the device restarts with the fiber cable already connected to the SFP module, it enters a reboot loop.

The only workaround I've found is to disconnect the fiber cable, wait for the device to boot up fully, and then reconnect the fiber. This is not practical, especially in environments where physical access to the device is limited.

Is there any version of the firmware that addresses this issue or any workaround that doesn't require manually unplugging the fiber cable after every power failure?

Thanks in advance!

Anime4000, 2025-04-13:
What stick are you using? also what router do you use

AntoniooDidier, 2025-04-13:
Hello! Iam using the ODI DFP-34X-2C3 with Unifi Cloud Gateway Fiber 😁