Anime4000/RTL960x #396: Can't get DHCP address from VoIP VLAN

lifespirit, 2025-04-17:
Hello.
I have ODI DFP-34X-2C3 + Mikrotik and old Huawei HG8245 terminal. I copy base settings to OTU Web UI and I have O5 status. I get DHCP address from internet VLAN. But when I try get another DHCP address from VoIP VLAN I get nothing. I don't sure why that happend and don't understand what I need check. Can someone a little help me? Guide page or something...

OTU Status:
```
# omcicli mib get 84
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
VlanTagFilterData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
=================================
EntityID: 0x01
FilterTbl[0]: PRI 0,CFI 0, VID 10
FwdOp:  0x04
NumOfEntries: 1
=================================
=================================
EntityID: 0x02
FilterTbl[0]: PRI 0,CFI 0, VID 20
FwdOp:  0x04
NumOfEntries: 1
=================================
=================================
EntityID: 0x03
FilterTbl[0]: PRI 0,CFI 0, VID 30
FwdOp:  0x04
NumOfEntries: 1
=================================
=================================
EntityID: 0x04
FilterTbl[0]: PRI 0,CFI 0, VID 40
FwdOp:  0x04
NumOfEntries: 1
=================================
=================================
EntityID: 0x05
FilterTbl[0]: PRI 0,CFI 0, VID 50
FwdOp:  0x04
NumOfEntries: 1
=================================
=================================
EntityID: 0x06
FilterTbl[0]: PRI 0,CFI 0, VID 60
FwdOp:  0x04
NumOfEntries: 1
=================================
=================================
EntityID: 0x07
FilterTbl[0]: PRI 0,CFI 0, VID 30
FilterTbl[2]: PRI 0,CFI 0, VID 60
FwdOp:  0x04
NumOfEntries: 2
=================================
=================================
EntityID: 0x08
FilterTbl[0]: PRI 0,CFI 0, VID 10
FwdOp:  0x04
NumOfEntries: 1
=================================```
I think VoIP VLAN is 10 and don't understand why what write twice.

Anime4000, 2025-04-18:
show `omcicli mib get 171`

this tell where VLAN goes where, eg: LAN1, LAN2, or TEL (VETH) or TR069 (VETH)

lifespirit, 2025-04-19:
Thanks for answer.

# omcicli mib get 171
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
ExtVlanTagOperCfgData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX

That's all.
On my HG8245 VoIP connect to tel port.

lifespirit, 2025-04-23:
So... I flashed M114 and add bridge VLAN 10 manual. With that VoIP work, but no internet. Can I add VLAN10 manually in M110?