Anime4000/RTL960x #401: dose my SFP support Vlans?

qatar2030, 2025-04-20:
Dear all and @Anime4000 

[image]

I have SFP Gpon from HSGQ-XPON-Stick with Firmware Version V1.1.8-240408 work with MikroTik RB5009UG+S+
I was wandering if I can configure IPTV and VoIP or not?
because my plant with ISP 1G internet including IPTV and VoIP and I replace Huawei Gpon terminal with SFP Gpon from HSGQ-XPON and I tried hard and repeatedly, but I did not succeed. I think it is almost impossible to do this.
Dose my SFP Gpon from HSGQ-XPON-Stick support Vlans?
is it possible to configure DATA, VOIP and IPTV together through SFP Gpon from HSGQ-XPON-Stick? if yes Please shear with me mikrotik configuration
with regards

ToTheCLI, 2025-04-22:
Yes supported, and also in Mikrotik RouterOS creating multiple VLANs on an Interface is possible, the issue you have with VoIP and IPTV is the config on devices after the router...

qatar2030, 2025-04-24:
yes I know and I couldn't find someone to help me

Anime4000, 2025-04-25:
For Huawei OLT, you might need change FwdOp to 0x00 or 0x02 for VLAN to pass

and your Mikrotik need to handle the VLAN and IPTV IGMP Broadcast

ohmantics, 2025-04-27:
Luleey also is shipping newer firmware, [241026](https://www.luleey.com/wp-content/uploads/resource/M110_sfp_LuLeey_241026.tar), with this same automatic VLAN feature. It's not clear yet to me if it works well enough, plus I'm not seeing any way to persist FwdOp changes.

qatar2030, 2025-05-04:
> change FwdOp to 0x00 or 0x02 for VLAN to pass

how?
send me command please

qatar2030, 2025-05-22:
I got it 
the command is" omcicli mib set 84 0 16 0x00 "
and the result is

[image]

would someone help me to configure voip on mikrotik please

Anime4000, 2025-05-22:
can you print ME 171, see how VLAN of VoIP `MePoints` to

qatar2030, 2025-05-23:
> print ME 171

# omcicli mib get 171
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
ExtVlanTagOperCfgData
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
 I already flash HSGQ-XPON-Stick with firmware M110_sfp_ODI_240202.tar