Anime4000/RTL960x #410: Boot stick without fiber connected?

marvokurven, 2025-05-23:
Has anyone found a way to make these sticks boot without a working fiber connection, or at least have an idea how such a thing could be accomplished?
i've read a few forum posts that suggests the sticks will in fact boot without fiber on a narrow selection of Mikrotik hardware with RouterOS 7 and above for some reason, so it seems plausible somehow. 

The thing is, it's kinda awkward to disconnect my only internet connection just to be able to experiment with these things, in addition to the fact that to much experimenting could result in the ISP blocking the service without warning.
So being able to experiment safely without a fiber connection would be nice.

Anime4000, 2025-05-23:
buy media converter works without need fiber to be connected

Anime4000, 2025-05-23:
No, just find 2.5G Media Converter, those using RTL8221B for bridging

CpServiceSpb, 2025-05-24:
Some Keenetic, Mikrotik routers, some Dlink and Planet switches (not all) works fine without fiber connected toi the plauuged stick.
As I conversated with personel from my Dlink local office, it depends Link Fault Pass Through (LFP) functionality.

marvokurven, 2025-05-25:
Can this behaviour be disabled, preferably temporarily, or is it implemented at a hardware level in media converters?