Anime4000/RTL960x #426: PPPoE doesn't works on DFP-34X-2C2, only on original FW

Gradius2, 2025-08-21:
Hi,

I'd tried every M110 firmware from here:
https://github.com/Anime4000/RTL960x/tree/main/Firmware/DFP-34X-2C2

So far, **ONLY** M110_sfp_ODI_210702.tar works fine.

I use Movistar here (Telefonica in Chile), and it uses PPPoE, besides bridge mode.

I don't know why, but every other firmware, refuses to works fine, it refuses to go into "bridge mode" (it seems) and works under PPPoE (my pfSense is doing the bridge and PPPoE part).

I want to try other FW mainly to keep my DFP in 2,5Gbps mode.

Any help?

Gradius2, 2025-09-01:
Not an issue, but I want the current modified webgui, looks so good and polite.  Where can I download ?

Anime4000, 2025-09-01:
> I want the current modified webgui, looks so good and polite. Where can I download ?

it's not available, as it contain a lot of ISP proprietary stuff, quite risk to release it

I am sorry